jackw01/led-control #1: Flask App cannot find index.html

Duwacell, 2019-02-26:
Hello,

I would like to try your tool for my LEDs connected to my Pi.

i did `python3 setup.py install` installation completed well

when i call for ledcontrol the app seams to lunch well but when i reach my pi IP i get this error

```
Loaded saved settings from /home/pi/led-control/ledcontrol.json
 * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
[2019-02-26 22:06:11,176] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/Flask-1.0.2-py3.5.egg/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.5/dist-packages/Flask-1.0.2-py3.5.egg/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.5/dist-packages/Flask-1.0.2-py3.5.egg/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python3.5/dist-packages/Flask-1.0.2-py3.5.egg/flask/_compat.py", line 35, in reraise
    raise value
  File "/usr/local/lib/python3.5/dist-packages/Flask-1.0.2-py3.5.egg/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.5/dist-packages/Flask-1.0.2-py3.5.egg/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/local/lib/python3.5/dist-packages/led_control-1.0.0-py3.5.egg/ledcontrol/app.py", line 80, in index
    colors=animation_controller.colors)
  File "/usr/local/lib/python3.5/dist-packages/Flask-1.0.2-py3.5.egg/flask/templating.py", line 134, in render_template
    return _render(ctx.app.jinja_env.get_or_select_template(template_name_or_list),
  File "/usr/local/lib/python3.5/dist-packages/Jinja2-2.10-py3.5.egg/jinja2/environment.py", line 869, in get_or_select_template
    return self.get_template(template_name_or_list, parent, globals)
  File "/usr/local/lib/python3.5/dist-packages/Jinja2-2.10-py3.5.egg/jinja2/environment.py", line 830, in get_template
    return self._load_template(name, self.make_globals(globals))
  File "/usr/local/lib/python3.5/dist-packages/Jinja2-2.10-py3.5.egg/jinja2/environment.py", line 804, in _load_template
    template = self.loader.load(self, name, globals)
  File "/usr/local/lib/python3.5/dist-packages/Jinja2-2.10-py3.5.egg/jinja2/loaders.py", line 113, in load
    source, filename, uptodate = self.get_source(environment, name)
  File "/usr/local/lib/python3.5/dist-packages/Flask-1.0.2-py3.5.egg/flask/templating.py", line 58, in get_source
    return self._get_source_fast(environment, template)
  File "/usr/local/lib/python3.5/dist-packages/Flask-1.0.2-py3.5.egg/flask/templating.py", line 83, in _get_source_fast
    return loader.get_source(environment, template)
  File "/usr/local/lib/python3.5/dist-packages/Jinja2-2.10-py3.5.egg/jinja2/loaders.py", line 171, in get_source
    f = open_if_exists(filename)
  File "/usr/local/lib/python3.5/dist-packages/Jinja2-2.10-py3.5.egg/jinja2/utils.py", line 154, in open_if_exists
    return open(filename, mode)
NotADirectoryError: [Errno 20] Not a directory: '/usr/local/lib/python3.5/dist-packages/led_control-1.0.0-py3.5.egg/ledcontrol/templates/index.html'
192.168.1.81 - - [26/Feb/2019 22:06:11] "GET / HTTP/1.1" 500 -
```

Do you have any idea how can i fix it ? 

Thanks for your time, Romain

jackw01, 2019-02-27:
Have you checked to see if index.html is actually inside `/usr/local/lib/python3.5/dist-packages/led_control-1.0.0-py3.5.egg/ledcontrol/templates/`?

Duwacell, 2019-02-27:
nop, there is no folders templates or static inside

jackw01, 2019-02-27:
Try cloning the repository and copying the templates and static folders from there, I will try to fix setup.py to copy them automatically

closed 2020-01-26 by jackw01