jackw01/led-control #2: Fresh install isn't working

glitch003, 2020-02-02:
Hi, thanks for making this awesome package!  

I just did a fresh install on a brand new Raspbian installation and I'm getting the error below.  Any ideas?

`pi@raspberrypi:~/led-control $ sudo ledcontrol --led_count 15
/usr/local/lib/python3.7/dist-packages/_rpi_ws281x.cpython-37m-arm-linux-gnueabihf.so
Traceback (most recent call last):
  File "/usr/local/bin/ledcontrol", line 11, in <module>
    load_entry_point('led-control', 'console_scripts', 'ledcontrol')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/led-control/ledcontrol/__init__.py", line 7, in <module>
    from ledcontrol.app import create_app
  File "/home/pi/led-control/ledcontrol/app.py", line 10, in <module>
    from ledcontrol.animationcontroller import AnimationController
  File "/home/pi/led-control/ledcontrol/animationcontroller.py", line 12, in <module>
    import ledcontrol.animationpatterns as patterns
  File "/home/pi/led-control/ledcontrol/animationpatterns.py", line 8, in <module>
    import ledcontrol.rpi_ws281x as rpi_ws281x
  File "/home/pi/led-control/ledcontrol/rpi_ws281x/__init__.py", line 19, in <module>
    blackbody_to_rgb = rpi_ws281x.blackbody_to_rgb
AttributeError: module '_rpi_ws281x' has no attribute 'blackbody_to_rgb'`

jackw01, 2020-02-07:
I recently had an issue doing a fresh install of this and changed `ledcontrol/rpi_ws281x/__init__.py` to fix it, but I'm not sure why the fix was necessary or why it even fixed anything.

Can you try replacing the contents of `/home/pi/led-control/ledcontrol/rpi_ws281x/__init__.py` with

```python
# led-control WS2812B LED Controller Server
# Copyright 2019 jackw01. Released under the MIT License (see LICENSE for details).

from .rpi_ws281x import PixelStrip, Color
from .lib.rpi_ws281x import (
    blackbody_to_rgb,
    blackbody_correction_rgb,
    wave_pulse,
    wave_triangle,      
    wave_sine,
    wave_cubic,
    plasma_sines,       
    plasma_sines_octave,
    perlin_noise_3d,    
)

__version__ = '4.2.2'
```

If this doesn't work, the issue is likely something with the build process for the C extension module.

m-seker, 2020-07-17:
I get the following if I replace the contents of __init__.py:
```
/usr/local/lib/python3.7/dist-packages/_rpi_ws281x.cpython-37m-arm-linux-gnueabihf.so
Traceback (most recent call last):
  File "/usr/local/bin/ledcontrol", line 11, in <module>
    load_entry_point('led-control', 'console_scripts', 'ledcontrol')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/led-control/ledcontrol/__init__.py", line 7, in <module>
    from ledcontrol.app import create_app
  File "/home/pi/led-control/ledcontrol/app.py", line 10, in <module>
    from ledcontrol.animationcontroller import AnimationController
  File "/home/pi/led-control/ledcontrol/animationcontroller.py", line 12, in <module>
    import ledcontrol.animationpatterns as patterns
  File "/home/pi/led-control/ledcontrol/animationpatterns.py", line 8, in <module>
    import ledcontrol.rpi_ws281x as rpi_ws281x
  File "/home/pi/led-control/ledcontrol/rpi_ws281x/__init__.py", line 5, in <module>
    from .lib.rpi_ws281x import (
  File "/home/pi/led-control/ledcontrol/rpi_ws281x/lib/rpi_ws281x.py", line 255, in <module>
    class color_hsv(_object):
  File "/home/pi/led-control/ledcontrol/rpi_ws281x/lib/rpi_ws281x.py", line 261, in color_hsv
    __swig_setmethods__["hue"] = _rpi_ws281x.color_hsv_hue_set
AttributeError: module '_rpi_ws281x' has no attribute 'color_hsv_hue_set'

```

Any idea ?

WhiteyFord, 2020-08-05:
Great project mate ..... I'm having the exact same issue.

jackw01, 2020-08-09:
@WhiteyFord cannot reproduce, what version of Raspberry Pi OS are you running and are you following the exact instructions in README.md?

jackw01, 2020-09-10:
Got a chance to test this with a fresh install of the latest version of Raspberry Pi OS. It did not have any issues installing when I followed the directions in README.md.

closed 2020-09-10 by jackw01