jackw01/led-control #3: Systemd service

rabbit52068, 2020-07-04:
Hi jackw01, thanks for the great work!

The program works for me smoothly.
However, when I made it into a service, the web server works but the led didn't light up.
Would you mind to have a look?
Thanks in advance!

```
[Unit]
Description=LED Control
After=network.target

[Service]
ExecStart=/usr/local/bin/ledcontrol --led_count '2'
StandardOutput=inherit
StandardError=inherit

[Install]
WantedBy=multi-user.target
```

jackw01, 2020-07-09:
ledcontrol needs to be run as root for the LED DMA driver to work.

djmitho, 2020-07-13:
> ledcontrol needs to be run as root for the WS2812 DMA driver to work.

Hello jackw01!!! very good job!! i really like it
I am trying to start on boot but it's not working
As i know is needed to run as root but there is not working on crontab
can i do something for autostart?

javl, 2021-08-04:
Just for anyone wanting to do the same, this works for me:

```
[Unit]
Description=LED control
After=network.target

[Service]
ExecStart=/usr/local/bin/ledcontrol --led_count 289 --led_pixel_order RGB
WorkingDirectory=/home/pi
StandardOutput=inherit
StandardError=inherit
Restart=always
User=root

[Install]
WantedBy=multi-user.target
```
Save as ```/etc/systemd/system/ledcontrol.service``` and install with ```sudo systemctl enable ledcontrol.service```