jackw01/led-control #4: AttributeError: module '_rpi_ws281x' has no attribute 'blackbody_to_rgb'

ChristopherLBruce, 2020-10-24:
Installed on Raspberry Pi following steps in readme.
sudo ledcontrol --led_count 384 --port 5000

Traceback (most recent call last):
  File "/usr/local/bin/ledcontrol", line 11, in <module>
    load_entry_point('led-control', 'console_scripts', 'ledcontrol')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/Python/led-control/ledcontrol/__init__.py", line 7, in <module>
    from ledcontrol.app import create_app
  File "/home/pi/Python/led-control/ledcontrol/app.py", line 10, in <module>
    from ledcontrol.animationcontroller import AnimationController
  File "/home/pi/Python/led-control/ledcontrol/animationcontroller.py", line 12, in <module>
    import ledcontrol.animationpatterns as animpatterns
  File "/home/pi/Python/led-control/ledcontrol/animationpatterns.py", line 7, in <module>
    import ledcontrol.rpi_ws281x as rpi_ws281x
  File "/home/pi/Python/led-control/ledcontrol/rpi_ws281x/__init__.py", line 19, in <module>
    blackbody_to_rgb = rpi_ws281x.blackbody_to_rgb
AttributeError: module '_rpi_ws281x' has no attribute 'blackbody_to_rgb'

jackw01, 2020-10-27:
It doesn't look like the C bindings compiled correctly. Were there any errors during the installation process? I recently tested this on a clean installation of Raspberry Pi OS (2020-08-20 version) and it did not have any issues.

ChristopherLBruce, 2020-11-01:
Was able to get this to work on fresh install of Raspbian

closed 2020-11-01 by ChristopherLBruce