jackw01/led-control #5: AttributeError: module '_rpi_ws281x' has no attribute 'blackbody_to_rgb'

ZehelSitchel, 2020-11-20:
Same issue as #4 . This is the output when running `sudo python3 setup.py develop`
Fresh install of raspbian. `2020-08-20-raspios-buster-armhf-lite.img`
```
running develop
running egg_info
writing led_control.egg-info/PKG-INFO
writing dependency_links to led_control.egg-info/dependency_links.txt
writing entry points to led_control.egg-info/entry_points.txt
writing requirements to led_control.egg-info/requires.txt
writing top-level names to led_control.egg-info/top_level.txt
reading manifest file 'led_control.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'led_control.egg-info/SOURCES.txt'
running build_ext
building '_rpi_ws281x' extension
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Iledcontrol/rpi_ws281x/lib -I/usr/include/python3.7m -c ledcontrol/rpi_ws281x/lib/rpi_ws281x_wrap.c -o build/temp.linux-armv6l-3.7/ledcontrol/rpi_ws281x/lib/rpi_ws281x_wrap.o
In file included from ledcontrol/rpi_ws281x/lib/rpi_ws281x_wrap.c:3101:
ledcontrol/rpi_ws281x/lib/led_render.h: In function ‘unpack_rgb’:
ledcontrol/rpi_ws281x/lib/led_render.h:31:10: warning: missing braces around initializer [-Wmissing-braces]
   return (color_rgb){ r, g, b };
          ^
                       --------
                       {{{r}, {g}, {b }}}
ledcontrol/rpi_ws281x/lib/led_render.h: In function ‘blackbody_to_rgb’:
ledcontrol/rpi_ws281x/lib/led_render.h:79:10: warning: missing braces around initializer [-Wmissing-braces]
   return (color_rgb_float){r_out, g_out, b_out};
          ^
                            {{{  } {    } {    }}}
ledcontrol/rpi_ws281x/lib/led_render.h: In function ‘blackbody_correction_rgb’:
ledcontrol/rpi_ws281x/lib/led_render.h:84:10: warning: missing braces around initializer [-Wmissing-braces]
   return (color_rgb_float){bb.r * rgb.r, bb.g * rgb.g, bb.b * rgb.b};
          ^
                            {{{         } {           } {           }}}
At top level:
ledcontrol/rpi_ws281x/lib/rpi_ws281x_wrap.c:3055:12: warning: ‘convert_iarray_32’ defined but not used [-Wunused-function]
 static int convert_iarray_32(PyObject *input, uint32_t *ptr, int size) {
            ^~~~~~~~~~~~~~~~~
ledcontrol/rpi_ws281x/lib/rpi_ws281x_wrap.c: In function ‘_wrap_ws2811_hsv_render_array_float’:
ledcontrol/rpi_ws281x/lib/rpi_ws281x_wrap.c:7054:15: warning: ‘arg3’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     if (arg3) free(arg3);
               ^~~~~~~~~~
ledcontrol/rpi_ws281x/lib/rpi_ws281x_wrap.c: In function ‘_wrap_ws2811_rgb_render_array_float’:
ledcontrol/rpi_ws281x/lib/rpi_ws281x_wrap.c:7161:15: warning: ‘arg3’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     if (arg3) free(arg3);
               ^~~~~~~~~~
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv6l-3.7/ledcontrol/rpi_ws281x/lib/rpi_ws281x_wrap.o -Lledcontrol/rpi_ws281x/lib/c/rpi_ws281x/ -lws2811 -o build/lib.linux-armv6l-3.7/_rpi_ws281x.cpython-37m-arm-linux-gnueabihf.so
copying build/lib.linux-armv6l-3.7/_rpi_ws281x.cpython-37m-arm-linux-gnueabihf.so ->
Creating /usr/local/lib/python3.7/dist-packages/led-control.egg-link (link to .)
led-control 1.0.0 is already the active version in easy-install.pth
Installing ledcontrol script to /usr/local/bin

Installed /home/pi/led-control
Processing dependencies for led-control==1.0.0
Searching for ujson==4.0.1
Best match: ujson 4.0.1
Processing ujson-4.0.1-py3.7-linux-armv6l.egg
ujson 4.0.1 is already the active version in easy-install.pth

Using /home/pi/led-control/.eggs/ujson-4.0.1-py3.7-linux-armv6l.egg
Searching for RestrictedPython==5.1
Best match: RestrictedPython 5.1
Processing RestrictedPython-5.1-py3.7.egg
RestrictedPython 5.1 is already the active version in easy-install.pth

Using /home/pi/led-control/.eggs/RestrictedPython-5.1-py3.7.egg
Searching for Flask==1.1.2
Best match: Flask 1.1.2
Processing Flask-1.1.2-py3.7.egg
Flask 1.1.2 is already the active version in easy-install.pth
Installing flask script to /usr/local/bin

Using /home/pi/led-control/.eggs/Flask-1.1.2-py3.7.egg
Searching for recordclass==0.14.3
Best match: recordclass 0.14.3
Processing recordclass-0.14.3-py3.7-linux-armv6l.egg
recordclass 0.14.3 is already the active version in easy-install.pth

Using /home/pi/led-control/.eggs/recordclass-0.14.3-py3.7-linux-armv6l.egg
Searching for itsdangerous==1.1.0
Best match: itsdangerous 1.1.0
Processing itsdangerous-1.1.0-py3.7.egg
itsdangerous 1.1.0 is already the active version in easy-install.pth

Using /home/pi/led-control/.eggs/itsdangerous-1.1.0-py3.7.egg
Searching for click==7.1.2
Best match: click 7.1.2
Processing click-7.1.2-py3.7.egg
click 7.1.2 is already the active version in easy-install.pth

Using /home/pi/led-control/.eggs/click-7.1.2-py3.7.egg
Searching for Werkzeug==1.0.1
Best match: Werkzeug 1.0.1
Processing Werkzeug-1.0.1-py3.7.egg
Werkzeug 1.0.1 is already the active version in easy-install.pth

Using /home/pi/led-control/.eggs/Werkzeug-1.0.1-py3.7.egg
Searching for Jinja2==2.11.2
Best match: Jinja2 2.11.2
Processing Jinja2-2.11.2-py3.7.egg
Jinja2 2.11.2 is already the active version in easy-install.pth

Using /home/pi/led-control/.eggs/Jinja2-2.11.2-py3.7.egg
Searching for MarkupSafe==1.1.1
Best match: MarkupSafe 1.1.1
Processing MarkupSafe-1.1.1-py3.7-linux-armv6l.egg
MarkupSafe 1.1.1 is already the active version in easy-install.pth

Using /home/pi/led-control/.eggs/MarkupSafe-1.1.1-py3.7-linux-armv6l.egg
Finished processing dependencies for led-control==1.0.0
```

jackw01, 2020-11-21:
Since I have been unable to reproduce this issue and the output from `sudo python3 setup.py develop` is identical to the output I get, can you send the output when running `sudo ledcontrol --led_count 150` and upload a tar.gz or zip archive of the whole `led-control` folder?

ZehelSitchel, 2020-11-22:
Absolutely! Thank you for looking at it. I also added the full terminal output to the tar.gz from fresh OS install to installing LED Control.

```
pi@raspberrypi:~/led-control $ sudo ledcontrol --led_count 150
/usr/local/lib/python3.7/dist-packages/_rpi_ws281x.cpython-37m-arm-linux-gnueabihf.so
Traceback (most recent call last):
  File "/usr/local/bin/ledcontrol", line 11, in <module>
    load_entry_point('led-control', 'console_scripts', 'ledcontrol')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/led-control/ledcontrol/__init__.py", line 7, in <module>
    from ledcontrol.app import create_app
  File "/home/pi/led-control/ledcontrol/app.py", line 10, in <module>
    from ledcontrol.animationcontroller import AnimationController
  File "/home/pi/led-control/ledcontrol/animationcontroller.py", line 12, in <module>
    import ledcontrol.animationpatterns as animpatterns
  File "/home/pi/led-control/ledcontrol/animationpatterns.py", line 7, in <module>
    import ledcontrol.rpi_ws281x as rpi_ws281x
  File "/home/pi/led-control/ledcontrol/rpi_ws281x/__init__.py", line 19, in <module>
    blackbody_to_rgb = rpi_ws281x.blackbody_to_rgb
AttributeError: module '_rpi_ws281x' has no attribute 'blackbody_to_rgb'
pi@raspberrypi:~/led-control $
```

[Raspberry Pi.tar.gz](https://github.com/jackw01/led-control/files/5579924/Raspberry.Pi.tar.gz)

jackw01, 2020-11-22:
It seems like this error was being caused by installing the rpi_ws281x library through pip as suggested in the Adafruit guide. I changed the name of the LEDControl extension module so Python won't confuse it with rpi_ws281x and also simplified the install process so only `sudo python3 setup.py develop` needs to be run. Let me know if it works.

ZehelSitchel, 2020-11-23:
Worked perfectly. Thank you again for the excellent program and your help! Time for me to integrate the LED strip and Pi into my desk. 😄👍

closed 2020-11-23 by jackw01