jackw01/led-control #6: Everything is seemingly working, but I'm getting nothing.

Pure-Audio, 2020-12-11:
I'm working on an LED Pickguard, and just testing with 1 SK6812 (RGBW) before I order the PCB to mount them all on, and everything seems to be working, getting all the way to the web control, yet the LED still won't do anything.

Not sure if this is even useful, but I'm measuring the power supply voltage fine, but not getting anything when measuring the signal to ground. (I'm running it through a step up converter which I have set to 5v.)

I'm basically brand new to this stuff, so don't really know what info is useful.

jackw01, 2020-12-13:
- What model Raspberry Pi are you using?
- What OS/version are you running on the Pi?
- What arguments are you running ledcontrol with (especially --led_pin, --led_data_rate, --led_dma_channel)?
- Are you running ledcontrol as root?
- Are any other programs using the Raspberry Pi's onboard audio?
- Are you sure the LED is connected to a valid pin? The default is GPIO/BCM pin 18 (pin 12 on the header). Not all GPIO pins can be used, see [here](https://github.com/jgarff/rpi_ws281x) for more information.

Pure-Audio, 2020-12-13:
RPi 4 Model B 2GB

Latest Version as of the 2nd of December

Was running
Sudo ledcontrol --led_count 1 --led_pixel_order grbw

Yes

Do I need to disable the analogue audio? I'm not running anything else or anything in the jack.

Yeah I'm using the default gpio 18

[image]

jackw01, 2020-12-13:
I think the step up converter is the problem, the one you are using is intended for power supply usage and not for boosting an 800kHz low-current signal. A logic level converter is what you would want to use for an application like this.

From my experience, boosting the 3.3V output from the Raspberry Pi is not necessary if the first LED is relatively close to the Raspberry Pi, so try connecting GPIO 18 directly to the LED data input (with the LED still being powered from the 5V rail). The only time I've had issues was when the LED data line was connected through a 5 meter cable.

Pure-Audio, 2020-12-14:
Cool I'll try ythat and let you know how it's goes.

closed 2020-12-20 by jackw01