jackw01/led-control #7: Random Flicker

rev-57, 2020-12-21:
So I setup a strip of SK6812 LEDs (2m, 120 LEDs in total) with a Raspberry Pi 4 Model B. The software works perfectly with the LEDs except for at random intervals, the LEDs will flicker.

For example, when left on "Solid Colour" on red (255,0,0), the LEDs will randomly flicker to white at full brightness. Sometimes it will be frequent (3/4 times in 10 seconds) while sometimes it can go half a minute or longer without flickering.

I'm uncertain as to what could be causing this. I've tried changing the FPS to no avail (default of 60 to 30 then to 24). I've tried setting up the circuit with a level converter and without one. The program and lights work on both occasions but the flickering still happens. I've tried switching DMA channels, the problem still persists.

Any help would be appreciated!

rev-57, 2020-12-21:
After shutting off my Pi for a while and then booting it and the program up again, there seems to be no flickering. I'm gonna put it down to either me fiddling with the settings erratically initially caused it to have a tantrum or just using it during a bad moment.

For now, the issue is resolved.

closed 2020-12-21 by rev-57

jackw01, 2020-12-22:
Sometimes flickering can be caused by a defective LED, an inadequate power supply, or (most of the time) a loose data connection. 

- If one LED fails and no longer transmits data properly, all LEDs downstream from that one will display the wrong color or flicker. 
- Reduce the brightness and see if the flickering stops, if so the power supply may not be able to provide enough current to run the LEDs at full brightness.
- Try jiggling the LED data line and ground connections between the LED strip and Raspberry Pi to see if it causes flickering or stops the LEDs from flickering.