jackw01/led-control #9: Optimization: Stop LED Update Timer if not required

nils, 2020-12-24:
Hi Jack,

this is another suggestion/idea that I'd like to discuss with you:

Is it possible to "stop" led-control from the web interface somehow? Currently I set  "Master Brightness" to zero if I don't need the LEDs, but I see that the timer for updating the LEDs is probably still running, because ledcontrol continues to use 30-50% CPU of an RPI 3. What do you think about an optimization there for "static configurations" where the LEDs don't animate, but have a static color over time?

- There could be an explicit on/off toggle on the web interface, that stops the timer. I assume the LEDs will stay in whatever state they are when you stop the timer, right?
- Alternatively, could we stop the timer as soon as "master brightness" is turned to zero? However, this would not support the "static configurations" usecase mentioned above, but only be a way to turn off.
- Pattern functions could return an optional boolean giving information about whether the pattern function is "static", i.e. will always return the same values independent of t/dt? Then the timer could also be stopped as soon as the function was executed once for each LED.

What do you think about these ideas? I could also give it a try to implement them, however I'm not exactly used to Python coding. Maybe you'd have less work doing it yourself than reviewing my stuff :smirk:

Best Regards,
Nils

@jackw01

jackw01, 2020-12-24:
Implemented in 3b7bb37d668e934553726b7b771768219618cb08: The next animation frame will not be rendered if a static pattern is being displayed (solid color or the newly added gradient pattern), if brightness is set to 0, or if primary speed is set to 0. 

CPU usage when LEDs are static is reduced from ~70% to ~3% (Raspberry Pi Zero, RGB Octave Plasma animation, 150 RGBW LEDs at 60FPS)

closed 2020-12-24 by jackw01

nils, 2020-12-25:
Wow, this was fast! :smile: Thanks a lot!

nils, 2020-12-25:
Hmm I think I found a bug/unexpected behavior testing this: Changing e.g. "Master Brightness" or the palette color in "Solid Color" mode is not immediately visible any more... It would need to execute even static patterns once after these inputs have been changed.

jackw01, 2020-12-25:
Fixed in f35f583b0db3a2f5cca72611ccd59d05e0352f40