jackw01/led-control #10: Thank you for working

nils, 2021-01-24:
Hi @jackw01,

I just wanted to let you know that your recent changes (commit https://github.com/jackw01/led-control/commit/bc3e76fc16f1dea6363582ad40b22fa55e816d7f) concerning color correction are appreciated!

For me, a white gradient has changed from this:
[image]
to this on the most recent version:
[image]

I'm using `--led_color_correction #FFFFFF` and color temperature 6600K by the way, I think that's better for white gradients, right?

However, depending on the pattern, the color correction might still be wanted... Maybe the pattern could control that itself by returning new color formats in addition to `rgb` and `hsv` (e.g. `rgbraw` and `hsvraw`?) where the `*raw` types skip color correction?

Best Regards,
Nils

jackw01, 2021-01-25:
Are you using an RGBW LED strip? If you are, the color correction and color temperature will not affect how white looks since they are applied to the RGB channels only after the white pixel value is calculated.

nils, 2021-01-25:
Thanks for the clarification! Yes I'm indeed using an SK6812 RGBW strip. 

Good to know that white colors are not affected at all in this case 👍 So that's what you fixed in https://github.com/jackw01/led-control/commit/bc3e76fc16f1dea6363582ad40b22fa55e816d7f!

closed 2021-01-25 by nils