jackw01/led-control #14: Error on a newly installed raspebrry pi 4

springjools, 2021-08-19:
pi@uranian:~ $ /usr/local/bin/ledcontrol -h
Traceback (most recent call last):
  File "/usr/local/bin/ledcontrol", line 11, in <module>
    load_entry_point('led-control', 'console_scripts', 'ledcontrol')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/led-control/ledcontrol/__init__.py", line 7, in <module>
    from ledcontrol.app import create_app
  File "/home/pi/led-control/ledcontrol/app.py", line 24, in <module>
    '', '', False,
  File "/home/pi/led-control/.eggs/recordclass-0.15.1-py3.7-linux-aarch64.egg/recordclass/recordclass.py", line 67, in recordclass
    fast_new=fast_new, gc=False)
  File "/home/pi/led-control/.eggs/recordclass-0.15.1-py3.7-linux-aarch64.egg/recordclass/dataclass.py", line 152, in make_dataclass
    cls = datatype(typename, bases, ns, gc=gc, fast_new=fast_new)
  File "/home/pi/led-control/.eggs/recordclass-0.15.1-py3.7-linux-aarch64.egg/recordclass/datatype.py", line 154, in __new__
    __new__ = _make_new_function(typename, fields, defaults, annotations, use_dict)
  File "/home/pi/led-control/.eggs/recordclass-0.15.1-py3.7-linux-aarch64.egg/recordclass/datatype.py", line 257, in _make_new_function
    code = compile(new_func_def, "", "exec")
  File "", line 3
    'Create new instance: FormItem(control='range', key='None', type=None, min=0, max=1, step=0.01, options=[], val=0, label='', unit='', hide=False)'

springjools, 2021-08-19:
Using raspberry pi 4 with 64-bit version:
**Linux uranian 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux**

jackw01, 2021-08-19:
That doesn't look like a complete Python error message. There should be a line at the end with the name of the exception that was thrown.

jackw01, 2021-08-20:
Try again with the latest commit, it should work.

closed 2021-08-23 by jackw01