jackw01/led-control #16: help troubleshooting random colors

ohthehugemanatee, 2021-09-25:
Thank you for all your work on this project! 

I'm having trouble getting [my SK6812W strip](https://www.amazon.de/gp/product/B079ZS6NQQ/ref=ppx_yo_dt_b_asin_title_o01_s05?ie=UTF8&psc=1) working properly with my Pi3B.  The strip is connected as described in the README, to GPIO18 with external power to the strip. When I run `sudo ledcontrol --led_count 300 --led_pixel_order GRBW`  the strip lights up and the flask interface looks lovely... but the led outputs on the strip are seemingly random.  "static color" shows up like this:

[image]

Any pattern set in the UI will send changes to the LED strip, but not in any recognizable pattern on the actual light strip. It is as if I am sending commands, but in the wrong protocol. 

How can I debug this further?

jackw01, 2021-09-25:
This is either a power supply issue, issue with the connection between the LEDs and the Raspberry Pi, bad LEDs, or an issue with the DMA configuration on the Pi.

- How are you powering the LEDs? 300 LEDs on one strip at full brightness will draw so much power that the voltage drop over the length of the LED strip may cause some of the LEDs to stop functioning properly. Try turning the brightness to 0.1 for testing to eliminate any power supply issues. If using a low quality power supply it may also be necessary to put a large (1000-4700uF) capacitor across the supply rails, close to the LED strip.
- Are the LED strip power supply ground and Raspberry Pi ground connected?
- Is the data line connected well and the wire as short as possible (try less than 200mm between the Raspberry Pi and LED strip)? If you are using a level shifter, try connecting it without a level shifter to see if it works.
- Do you have any other LEDs to test it with? What you are describing is the most common failure mode of addressable LED strips (usually caused by a wire bonding failure in one of the LED packages), but this is unlikely with new LEDs.
- Have you tried using other DMA channels or pins on the Raspberry Pi (see https://github.com/jgarff/rpi_ws281x)? DO NOT use DMA channel 5 on the Raspberry Pi 3B.

ohthehugemanatee, 2021-09-27:
Thank you! This is exactly the kind of helpful and informative response I was hoping for! I'm traveling on vacation this week, so I brought the pi and a 1m strip with me to continue poking at this. 

Brief answers:

- LEDs were powered by a molex connector on a separate ATX PSU. 5v with about 300A available. :)  I have a proper 5v powersource waiting for me when I get home. This might also be different when using a 1m strip powered by the pi 5v pin.
- yes, grounds were all connected. I used pin 9 on the pi for ground.
- The wire between the pi and the led strip is about 15cm. I can try clipping it shorter. I am not using a level shifter.
- Most of the time, no group of the LEDs are displaying a consistent color. But sometimes in among the periodic changes, a segment at the end or in the middle will go to the right color palette. Doesn't seem random, but I can't predict it.
- It's a vanilla install of raspberry pi OS... I could try using Pi OS lite to make sure nothing is using the sound driver, though. 
- no, haven't tried using other DMA channels. It seemed unnecessary. 

There's lots of great ideas in the jgarff repo you linked. I will try those too, and report back here.

jackw01, 2021-09-28:
I would definitely try powering everything through a single USB port (with the LEDs at low brightness, of course). IIRC, some older or low quality ATX power supplies cannot properly regulate their outputs without a fairly significant load and may sporadically shutdown and restart. If it's relatively new and high quality (well-known brand, 80+ rated, and UL/ETL listed) this shouldn't be a problem but any power supply issues can be eliminated by using a known stable power source like a 2+ amp USB charger.

If some of the LEDs do occasionally display the right colors, this can be a symptom of a loose/intermittent connection in the data line. Try bending/jiggling the wires, connectors, and even the LED strip itself and see if it affects anything. 

From my experience, 5V is not necessary on the data line unless you are driving the LEDs through a very long wire. Older WS2811 or WS2812 LEDs may have required 5 volts but I have used many WS2812B and SK6812 LED strips from various sources with 3.3V with no issues.

ohthehugemanatee, 2021-09-30:
W00t! It's working now at least in a limited setup. Here's what I did, for posterity:

- cut down to a 1m, 60LED strip
- power from pin 4, ground from pin 6, GPIO18 from pin 12
- used Raspberry Pi OS Lite, so I could be sure nothing else was using GPIO18 for sound etc

I was using a new ATX power supply, so my present theory is that it was something else competing for the GPIO18 pin. When I get back from holiday I'll try again with a full length strip and a "proper" 5v 60A power unit. (that's 8A per 60 pixel meter, should power a couple of 3-4m strips)

Closing this ticket now. Thank you again for your help, I hope this thread helps some future users. Now it's on to actually writing a few patterns...

closed 2021-09-30 by ohthehugemanatee

jackw01, 2021-09-30:
Tip: With a 60 amp power supply, a short circuit somewhere along the LED strip would likely not draw enough current to trip your power supply's over-current protection due to the resistance of the flexible PCB. This would be a pretty huge fire hazard if it ever happened, so I would recommend connecting sections of the strip that draw no more than ~10A to the power supply through their own independent wires with fuses or circuit breakers in series if you ever plan on running it unsupervised.