jackw01/led-control #27: Segment a string into sub-sections

uresu, 2021-11-21:
Is it possible to for example have LED 1-30 do one pattern and have 30-60 another... and so on?

jackw01, 2021-12-27:
This is not currently supported and it would require substantial modifications to the software to do this.

jackw01, 2022-02-06:
Currently under consideration as part of rewriting the web interface to use Vue. Will see how much it impacts performance when implementing it in the backend.

jackw01, 2022-02-10:
[image]

For now the UI will support adding, removing, renaming, and changing the LEDs spanned by groups. Each group has its own brightness/saturation/animation/palette settings and creating a preset will save settings for all currently existing groups. When loading a preset, saved settings for groups that no longer exist or have been deleted and recreated will be ignored. The default "main" group can be renamed but not deleted.

Future improvements could include somehow integrating pixel mappings into the web UI (whether through a text field to input JSON, python scripting to generate pixel mappings, or a full GUI) and then allowing each group to have a self-contained pixel mapping.

closed 2022-03-19 by jackw01