jackw01/led-control #28: sACN RGBW Overflow error

JosephAntony1, 2022-01-05:
```
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/led-control/.eggs/sacn-1.8.1-py3.9.egg/sacn/receiving/receiver_socket_udp.py", line 57, in receive_loop
    self._listener.on_data(raw_data, time.time())
  File "/home/pi/led-control/.eggs/sacn-1.8.1-py3.9.egg/sacn/receiving/receiver_handler.py", line 59, in on_data
    self.fire_callbacks_universe(tmp_packet)
  File "/home/pi/led-control/.eggs/sacn-1.8.1-py3.9.egg/sacn/receiving/receiver_handler.py", line 143, in fire_callbacks_universe
    self._listener.on_dmx_data_change(packet)
  File "/home/pi/led-control/.eggs/sacn-1.8.1-py3.9.egg/sacn/receiver.py", line 45, in on_dmx_data_change
    callback(packet)
  File "/home/pi/led-control/ledcontrol/animationcontroller.py", line 106, in _sacn_callback
    self.led_controller.set_all_pixels_rgb_float(
  File "/home/pi/led-control/ledcontrol/ledcontroller.py", line 93, in set_all_pixels_rgb_float
    driver.ws2811_rgb_render_array_float(self._leds, self._channel, pixels, len(pixels),
OverflowError: in method 'ws2811_rgb_render_array_float', argument 4 of type 'int'
```

[image]

When I try and turn sACN on on the web interface (with 300 rgbw leds) I get this overflow error which I believe makes the sACN server unable to receive any input, so the lights don't change when LedFx sends them a command

JosephAntony1, 2022-01-05:
I figured out the problem, [this line](https://github.com/jackw01/led-control/blob/904972b63b8d61e9a0913e430d72eef9e3a8e7f5/ledcontrol/animationcontroller.py#L107) is *3 for rgb instead of *4 for rgbw, if you could somehow add an if statement that'd be swell!

jackw01, 2022-01-05:
This is intentional because LEDFx and similar programs do not support RGBW (four-channel DMX) control. LEDControl only takes RGB input over sACN and the `ws2811_rgb_render_array_float` function automatically handles converting between color spaces if RGBW LEDs are being used. 

This error message indicates some issue with the data being sent over sACN that is causing an integer overflow. What are you using to send sACN input? 

Can you add `print(pixels)` as the first line of the function `def set_all_pixels_rgb_float(self, pixels, correction, saturation, brightness, gamma):` in ledcontroller.py to see what the actual pixel data being sent to the rendering code is?

JosephAntony1, 2022-01-05:
Sorry I'll respond to the above message in a minute, I think part of the problem is in when you truncate the incoming list at [this line](https://github.com/jackw01/led-control/blob/904972b63b8d61e9a0913e430d72eef9e3a8e7f5/ledcontrol/animationcontroller.py#L105), when I change it to 510, everything actually works perfectly

jackw01, 2022-01-05:
In that case it seems like you're running into an inherent limitation of DMX/sACN which is that you can only address 512 channels (up to 170 RGB LEDs). It looks like the LEDFx web interface allows you to enter a higher number of LEDs and doesn't perform any input checking, causing it to send malformed packets.

JosephAntony1, 2022-01-05:
Gotcha, that seems like it's going to be a solid sticking point for me, but thanks for your quick responses! learning a lot with this - do you want to cap that truncating variable at 510 though so other people don't run into this maybe?

shauneccles, 2022-01-05:
sACN can support 64k universes, with each universe having either 128 RGBW or 170 RGB pixels - the limiting factor within the universe is number of channels. 

Each Universe has 512 channels, each of which carries a value that corresponds to R, G, B (W) - a pixels data cannot span across universes, so as a sender LedFx breaks these up into groups of 170 pixels, and spins up a new sender universe for each group of pixels.

It appears that you're only looking at universe 1 in https://github.com/jackw01/led-control/blob/904972b63b8d61e9a0913e430d72eef9e3a8e7f5/ledcontrol/animationcontroller.py#L218

If you instead listen for multiple universes (you could listen for a new universe listener if you have >= 510 channels of pixel data in any one universe, but it seems a bit hacky), you should be able to pick up any number of pixels.

jrparadis, 2022-02-23:
I'm having the same issue, going below 170 LEDs works, but above that, sacn doesn't work.