jackw01/led-control #29: Is there a way to factor in the color temperature of the white LEDs?

JosephAntony1, 2022-01-13:
I can't quite figure out how the conversion of rgb -> rgbw when using the sACN mode is done so I wanted to know if there was some way to take into account the color temperature of the white LEDs in order to produce more accurate colors. Thanks!

jackw01, 2022-01-13:
Right now the white value is derived by finding the minimum of the red, green, and blue values which is then subtracted from the red, green, and blue channels. The white value is the square of this minimum value. After this, the red, green, and blue channels are multiplied by the `led_color_correction` values passed in through command line arguments and the color temperature set on the web interface. 

I think you could get pretty close by adjusting the `led_color_correction` value (for example, by decreasing the blue channel if the white LEDs are too cool). The best way (still not perfect) to implement this would be to have a second color correction value that represents the approximate color of the white LEDs in RGB, which is multiplied by the minimum value that gets subtracted from the RGB values. 

This still doesn't address several inherent issues in getting accurate colors with SK6812 RGBW LEDs: 
- The wavelengths of the RGB LEDs are unknown
- The exact color temperature/emission spectra of the white LEDs are unknown
- The brightness of the LEDs responds nonlinearly, which is hard to properly characterize and compensate for without specialized measuring equipment
- The blue LED excites the white LED's phosphor causing blue colors to appear desaturated
- All of these things may be inconsistent between suppliers of LEDs

This could all be addressed by taking objective measurements of various SK6812 LEDs from different suppliers and using the data to come up with a better color correction method, but I think at that point it would be less work and better overall to design a custom solution using known good LEDs and well-documented driver chips.

JosephAntony1, 2022-01-13:
Yeah that second method is more or less what I was envisioning, thanks for the info!

JosephAntony1, 2022-01-13:
Where in the code is the white value actually calculated?

jackw01, 2022-01-13:
https://github.com/jackw01/led-control/blob/master/ledcontrol/driver/led_render.h, lines 167-182 and 240-257

JosephAntony1, 2022-01-14:
Might be a dumb question but changing things in the led_render.h file doesn't seem to change the output at all...Does it get compiled or something?

jackw01, 2022-01-14:
Yes, the C bindings need to be generated and the code needs to be compiled by running `swig -python ./ledcontrol/driver/ledcontrol_rpi_ws281x_driver.i && sudo python3 setup.py develop` for changes to take effect.

WillCoulterKorrus, 2022-08-01:
Is it possible to allow the user to specify the r,g,b, and w channels directly instead of trying to map the rgbw onto rgbw with "saturation" and "global color temp"?

The code is impressive, but I'm having trouble finding where everything relates to eachother. I have access to the sophisticated measuring equipment mentioned in this thread, and I'm looking to control the strips at a more low-level. I surely wouldn't want to redo so much work you've done by creating my own implementation of the rpi_ws281x library.