jackw01/led-control #32: Remote management

Fitipalek, 2022-02-10:
Hi jack,
first of all great work, great project.
I consider myself as newbie with raspberry, especially with java. 
I bought materials and assembled version of two led corner lamps, something like this [](https://www.ledxx.com/led-floor-corner-lamp-en.html) .
I programmed some patterns in python for them but then saw this project and it was game changer.....

I have two questions:

- when i leave the program running for a while, web interface do not loads right up but it takes some time. Is that a power saving feature of raspberry pi zero 2 W or something else ?
- i would like to synchronise these two lamps to run these patterns at the same time, can you please point me to right direction on how to accomplish this ?

Thanks a lot

jackw01, 2022-02-10:
For the first question, my suspicion would be that it's caused by a memory leak in another program running on the Pi. I can personally leave LEDControl running for weeks or months at a time on a Raspberry Pi Zero 1.3 with no issues. Usually, the web interface will load faster over time due to browser caching.

1. Can you verify that you are using a stable version of LEDControl and not the latest commit (run `git checkout tags/v1.4.0`)?
2. Have you installed any other programs that run in the background?
3. Are you accessing files on the Raspberry Pi over the network using SMB when this happens?  
4. Is the Pi running a desktop environment, and are you using it when the problem occurs? 
5. Can you run the `top` command through ssh when the problem occurs to see what processes are running?

The best way to synchronize multiple lights is to have one Raspberry Pi and physically wire the data lines to each LED strip together. If you wire them in parallel (output from Raspberry Pi goes to the input of both LED strips), both will display the same colors. If you wire them in series (Raspberry Pi goes to the input of the first LED strip, output of first LED strip goes to input of second LED strip), the animation will be spread over both of them and you can control exactly how this happens by defining a pixel mapping as described in the README.

Fitipalek, 2022-02-14:
For the first problem: 
I bought these two raspberry for one purpose only, i installed raspberry pi OS lite 32 bit (can 64bit raspberry be problematic?), python 3.9 and then followed your instructions......then roll back to v1.4.0 as you recommended
Problem is difficult to replicate because after initial load web page works fine, nevertheless I copied output of top command while web interface was loading

PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
430 root      20   0   57084  21032   8192 S  22.1   4.8 152:05.57 ledcontrol
10674 pi        20   0   11256   3108   2632 R   1.0   0.7   0:06.23 top
10667 root      20   0       0      0      0 I   0.3   0.0   0:00.44 kworker/u8:1-brcmf_wq/mmc1:0001:1
    1 root      20   0   33700   8392   6652 S   0.0   1.9   0:11.54 systemd
    2 root      20   0       0      0      0 S   0.0   0.0   0:00.34 kthreadd
    3 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rcu_gp

May be, it is something with my old router.....

Regarding second problem i asked @javl in closed issues and his solution works good enough i guess. I like the portability of lamps and with remote management i can combine multiple devices in the future. I was wondering if you propose different solution :)

jackw01, 2022-02-15:
Please try reinstalling and see if the issue still exists, I just released v2.0.0 which uses a much faster web server. You will need to follow the installation instructions from step 1 since there is a new dependency.

Fitipalek, 2022-02-15:
This is definitely faster than the previous one, good job :), i'm not noticing any delay so far. 
I would switch step 3 and 4 thought and there is new problem which appeared on ipad, on pc and on android web elements are loading correctly but ipad loads only default html elements without the rest..... :) i tried safari and chrome...... but this is easy problem for you for sure :)

Thanks for your great work :)

jackw01, 2022-02-16:
Are you using any content blocking extensions on the iPad? What version of iOS is it running?

Fitipalek, 2022-02-16:
It is running IPadOS 15.3.1, im not using any content blocking extension deliberately..... i downloaded clean chrome on that ipad only yesterday to test if it is only safari problem or not..... :-/, may be they are on one update to late ?

jackw01, 2022-02-16:
Weird, it works on my iPhone also running iOS 15.3.1, which is the latest version. There should only be issues with iOS versions more than a few years out of date. All browsers on iOS will show the same thing because they are all required by Apple to use the standard Webkit-based iOS webview. I can't really debug this any further without being able to see what gets logged to the developer tools console on a device where this is happening. I know you can open developer tools on a website on an iOS device by connecting it to a Mac, and I think there may now be third-party browser extensions you can use to access some form of developer tools on the device itself.

Fitipalek, 2022-02-18:
[image]
[image]
IDK if this is what you wanted to post but i hope so :)
I tested the page also on IPhone and second IPad with the same results....

Thanks for help.

jackw01, 2022-02-19:
Can you send a screenshot of the Javascript console (the tab to the right of the one currently selected in the developer tools)? It looks like a script is not loading properly.

Fitipalek, 2022-02-19:
This is how it looks on tablet:
[image]
![photo_2022-02-19_10-57-19](https://user-images.githubusercontent.com/22022921/154796850-f11082f5-5679-4453-8258-d81a18cd0

And this on Mac:

[image]
772.jpg)

jackw01, 2022-02-19:
It looks like the scripts are loading properly on iOS but are not running at all - the console is not showing any of the messages that would be printed if the scripts run normally (like on the Mac) but is not showing any Javascript errors either. The two 404 NOT FOUND errors are normal.

Are you sure that Javascript is not disabled in Safari settings on the iPad?

Fitipalek, 2022-03-16:
Hi, im sure, it is not just safari...... :-/

jackw01, 2022-03-19:
Again, all web browsers on iOS use the same engine as Safari and some Safari settings affect the behavior of all browsers. The only explanation I can think of for this error is that scripts are either disabled or being blocked by a content blocking extension.

closed 2023-05-02 by jackw01