jackw01/led-control #33: v.2.0.0 Not setting UI elements after preset select

uresu, 2022-02-21:
[First of all, awesome work on v2.0.0, really enhances a great piece of work - do you want to see a video of my setup in action?]

The issue I have is that when I select a preset, the UI elements (select boxes etc) are not populated with the settings that they actually are.  I would expect the palettes and patterns for each section to be updated after a preset is selected.  I hope this is clear...

jackw01, 2022-02-21:
1. What web browser and version are you using?
2. Can you send a screenshot of what is reported in the Javascript console (press F12 or use the right click menu to open developer tools in your web browser, and select the Console tab)?

uresu, 2022-02-22:
Hi Jack, I'm using Chrome version 98.0.4758.102 (Official Build) (x86_64)

console shows :

Loading preset: All White
Store.js:80 Loading preset: Pride
Store.js:80 Loading preset: Forest
Store.js:80 Loading preset: All White
Store.js:80 Loading preset: Sunset

I actually think the view is being updated with the WRONG settings.  In the background the right settings are displayed on the LEDs but the UI is loading values from a different preset.

jackw01, 2022-02-22:
Can you send the config file (at /etc/ledcontrol.json by default)?

uresu, 2022-02-22:
```json
{
   "functions":{
      "0":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.0
      },
      "1":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.0
      },
      "100":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.2
      },
      "110":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.2
      },
      "112":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.2
      },
      "114":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.2
      },
      "120":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.2
      },
      "130":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.2
      },
      "140":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.1
      },
      "141":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.1
      },
      "150":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.05
      },
      "160":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.2
      },
      "161":{
         "default":true,
         "primary_scale":1.0,
         "primary_speed":0.2
      },
      "170":{
         "default":true,
[... 291 lines cut ...]
            "range_end":444,
            "range_start":355,
            "saturation":1,
            "scale":1,
            "speed":0.2
         },
         "main":{
            "brightness":1,
            "color_temp":6500,
            "function":1,
            "mapping":[
               
            ],
            "name":"Shelves",
            "palette":90,
            "range_end":171,
            "range_start":1,
            "saturation":1,
            "scale":1,
            "speed":0.2
         }
      },
      "sacn":0
   }
}

jackw01, 2022-03-19:
I have been using v2.0.0 on a daily basis for over a month and have tested with your configuration and haven't seen anything like this issue. If it is still a problem, can you possibly capture it in a screen recording?

closed 2023-07-10 by jackw01