jackw01/led-control #36: Random Lights

ascheier, 2022-03-15:
Well... I did have it working but I did something to my raspberry pi that required me to reinstall the OS. Anyway... I reinstalled the packages, following the READ ME and on the raspberry pi, it looks like it is working as expected. The lights, however, are not working. The first 6 LEDs are all different colors and the rest are off.
[image]
.

ascheier, 2022-03-17:
I fixed it. This could also fix an issue if people see the web interface but there are no lights on. Heres how I did it:
`sudo nano /boot/config.txt`
Change `# Enable audio (loads snd_bcm2835)
dtparam=audio=on`
to `# Enable audio (loads snd_bcm2835)
#dtparam=audio=on`

@jackw01 , I would consider mentioning this in the software setup section of the README.

closed 2022-03-17 by ascheier

ChrK2017, 2022-10-22:
Had the same problem, and now it works! Thanks!