jackw01/led-control #37: PackageNotFoundError: led-control

Macleykun, 2022-03-17:
I follow the build steps of this project but got this error that i couldn't find the led-control.
I did these steps:
```
sudo apt install scons swig libev-dev python3-dev python3-setuptools
git clone --recurse-submodules https://github.com/jackw01/led-control.git
cd led-control
git checkout tags/v2.0.0
sudo python3 setup.py develop
sudo ledcontrol --led_count
sudo ledcontrol --led_count 16 --led_pixel_order GRBW
```
I didn't had any errors during the build, only a few warnings about deprecated features and that cache wasn't being used.

Also, if i manage to get this to work, i would love to check out to make a container image for this project :) (maybe two for like a dev with ci/cd and a stable one like 2.0.0)

jackw01, 2022-03-19:
Where is the error?

Macleykun, 2022-03-19:
Terribly sorry for not including that! I've pasted the entire log here:
https://www.toptal.com/developers/hastebin/alomomakep.properties

I use raspberry pi OS 64-bit latest. I do know the ring works with the setup i have as the library you've updated (thanks for that!) has a demo which showed fully functionality :).

closed 2022-04-01 by Macleykun

cvrine, 2023-05-01:
I'm having the same error. After following all steps I get this error when running sudo led-control

Traceback (most recent call last):
  File "/usr/bin/ledcontrol", line 33, in <module>
    sys.exit(load_entry_point('led-control', 'console_scripts', 'ledcontrol')())
  File "/usr/bin/ledcontrol", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: led-control

jackw01, 2023-05-02:
@cvrine the command is `sudo ledcontrol` not `sudo led-control`

cvrine, 2023-05-02:
> @cvrine the command is `sudo ledcontrol` not `sudo led-control`

I ran sudo ledcontrol..

See full log
root@DietPi:~/led-control# sudo ledcontrol --led_count 60
Traceback (most recent call last):
  File "/usr/bin/ledcontrol", line 33, in <module>
    sys.exit(load_entry_point('led-control', 'console_scripts', 'ledcontrol')())
  File "/usr/bin/ledcontrol", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: led-control
root@DietPi:~/led-control#