jackw01/led-control #47: Alternative sigle board computers?

ThatOneVideoGamer, 2023-05-01:
Will this only work on a Raspberry pi? Those devices are extremely overpriced atm due to the shortage so I was wondering if it would work in an alternative like the orange pi

jackw01, 2023-05-02:
I have it on my todo list to make some modifications that will allow ledcontrol to run on any computer and control the LED strip through an RP2040 or Arduino connected via USB/serial.

jackw01, 2023-07-10:
Update: in progress. As of right now LEDControl can run on any computer and control the LED strip through a RP2040 microcontroller board connected over USB. The microcontroller is able to handle all the color conversions that were previously implemented in C, so nothing needs to be compiled and LEDControl should be able to run on any system with Python installed. Performance is as good or better than on Raspberry Pi and the CPU usage is negligible on a modern desktop computer.

closed 2023-07-13 by jackw01