jackw01/led-control #51: Failed building wheel for pyfastnoisesimd

WegnerDan, 2023-07-18:
I've tried following the steps under "Software Setup (Raspberry Pi)" on https://jackw01.github.io/led-control/

I have a raspberry pi zero w with most current raspberrypi os lite 32 bit image.

I get a lot of errors on step `sudo python3 setup.py develop`

[command output.txt](https://github.com/jackw01/led-control/files/12088356/command.output.txt)

jackw01, 2023-07-20:
Please try rebuilding with line 41 in setup.py commented out. The module that failed to build is only used for non-Raspberry Pi platforms, and if everything works properly without it, report the installation issue at the `pyfastnoisesimd` repository.

WegnerDan, 2023-07-22:
Build seems to be successful with line 41 commented out, but it does not run.

```
thedoginthewok@led-raspi:~/led-control $ sudo ledcontrol --led_count 10 --led_pixel_order GRBW
Traceback (most recent call last):
  File "/usr/local/bin/ledcontrol", line 33, in <module>
    sys.exit(load_entry_point('led-control', 'console_scripts', 'ledcontrol')())
  File "/usr/local/bin/ledcontrol", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.9/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/thedoginthewok/led-control/ledcontrol/__init__.py", line 5, in <module>
    from ledcontrol.app import create_app
  File "/home/thedoginthewok/led-control/ledcontrol/app.py", line 11, in <module>
    from ledcontrol.animationcontroller import AnimationController
  File "/home/thedoginthewok/led-control/ledcontrol/animationcontroller.py", line 14, in <module>
    import ledcontrol.animationfunctions as animfunctions
  File "/home/thedoginthewok/led-control/ledcontrol/animationfunctions.py", line 7, in <module>
    import ledcontrol.driver as driver
  File "/home/thedoginthewok/led-control/ledcontrol/driver/__init__.py", line 17, in <module>
    from . driver_non_raspberry_pi import *
  File "/home/thedoginthewok/led-control/ledcontrol/driver/driver_non_raspberry_pi.py", line 5, in <module>
    import pyfastnoisesimd as fns
ModuleNotFoundError: No module named 'pyfastnoisesimd'

```

<details>
<summary>Build Details</summary>

```
thedoginthewok@led-raspi:~/led-control $ sudo python3 setup.py develop
running develop
preinstall
running egg_info
creating led_control.egg-info
writing led_control.egg-info/PKG-INFO
writing dependency_links to led_control.egg-info/dependency_links.txt
writing entry points to led_control.egg-info/entry_points.txt
writing requirements to led_control.egg-info/requires.txt
writing top-level names to led_control.egg-info/top_level.txt
writing manifest file 'led_control.egg-info/SOURCES.txt'
reading manifest file 'led_control.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*' under directory 'ledcontrol/templates'
writing manifest file 'led_control.egg-info/SOURCES.txt'
running build_ext
Creating /usr/local/lib/python3.9/dist-packages/led-control.egg-link (link to .)
Adding led-control 2.0.0 to easy-install.pth file
Installing ledcontrol script to /usr/local/bin

Installed /home/thedoginthewok/led-control
Processing dependencies for led-control==2.0.0
Searching for sacn==1.9.0
Best match: sacn 1.9.0
Processing sacn-1.9.0-py3.9.egg
Adding sacn 1.9.0 to easy-install.pth file

Using /home/thedoginthewok/led-control/.eggs/sacn-1.9.0-py3.9.egg
Searching for pyOpenSSL==22.1.0
Best match: pyOpenSSL 22.1.0
Processing pyOpenSSL-22.1.0-py3.9.egg
[... 100 lines cut ...]
Adding async-timeout 4.0.2 to easy-install.pth file

Using /home/thedoginthewok/led-control/.eggs/async_timeout-4.0.2-py3.9.egg
Searching for zipp==3.16.2
Best match: zipp 3.16.2
Processing zipp-3.16.2-py3.9.egg
Adding zipp 3.16.2 to easy-install.pth file

Using /home/thedoginthewok/led-control/.eggs/zipp-3.16.2-py3.9.egg
Searching for MarkupSafe==2.1.3
Best match: MarkupSafe 2.1.3
Processing MarkupSafe-2.1.3-py3.9-linux-armv6l.egg
Adding MarkupSafe 2.1.3 to easy-install.pth file

Using /home/thedoginthewok/led-control/.eggs/MarkupSafe-2.1.3-py3.9-linux-armv6l.egg
Searching for pycparser==2.21
Best match: pycparser 2.21
Processing pycparser-2.21-py3.9.egg
Adding pycparser 2.21 to easy-install.pth file

Using /home/thedoginthewok/led-control/.eggs/pycparser-2.21-py3.9.egg
Finished processing dependencies for led-control==2.0.0
```

</details>

jackw01, 2023-07-23:
What does the output of the command `cat /sys/firmware/devicetree/base/model` say?

WegnerDan, 2023-07-23:
It looks a little weird, this is the exact output:
```
thedoginthewok@led-raspi:~ $ cat /sys/firmware/devicetree/base/model
Raspberry Pi Zero W Rev 1.1thedoginthewok@led-raspi:~ $
```

jackw01, 2023-07-24:
Not sure why this isn't working then. Can you try running these two lines of python code to see what the output is?
```
with io.open('/sys/firmware/devicetree/base/model', 'r') as m:
    print(m.read().lower())
```

For now, LEDControl should work if you replace the contents of `ledcontrol/driver/__init__.py` with `from _ledcontrol_rpi_ws281x_driver import *`.

commit 00afeee3 (2023-07-24): fix #51
closed 2023-07-24 by jackw01