jackw01/led-control #53: V.2.1.1 don't work with my raspberry version - setup.py don't work

xnathan97, 2023-10-02:
Hi, I have a raspberry pi 3b+. 
I tried to install the version v2.1.1. The hole install works until I ran the setup.py programm. I get this failures: "ERROR Failed buildung wheel for pyfastnoisesimd"

This failure comes, when I use v2.1.1. If i try it with v2.0.0 or lower it worked. 

Do you have any ideas?
Thank you!

jackw01, 2023-10-03:
Commenting out the line 
```
    'pyfastnoisesimd>=0.4.2',
```
in setup.py should fix this. This library is not used when running on Raspberry Pi anyways.

xnathan97, 2023-10-04:
Thx it's work. But when I try "sudo ledcontrol --led_count 100--led_pixel_order GRB" then I get the following message: ModuleNotFoundError: No module named '_ledcontrol_rpi_ws281x_driver'.
What can I do?

xnathan97, 2023-10-04:
Thx it works. 
But I get the message "ModuleNotFoundError: No module named '_ledcontrol_rpi_ws281x_driver'"
I follow all of your steps. What can I do?

teuwsenjung, 2023-10-04:
> Thx it works. 
> 
> But I get the message "ModuleNotFoundError: No module named '_ledcontrol_rpi_ws281x_driver'"
> 
> I follow all of your steps. What can I do?

I have the same problem...

jackw01, 2023-10-07:
Please try adding `import io` to the top of setup.py, rerun `python setup.py develop`, and see if this resolves the issue.

xnathan97, 2023-10-07:
Thx it works, but after the import io I got the message, Serial Module not found. 

I ran "sudo pip install pyserial" an adding "import serial" in the top of setup.py. 
Then I rerun "sudo python3 setup.py develop" and it works finally. 
Thx for your help.

commit 9feca2b0 (2023-10-26): fix #53
closed 2023-10-07 by xnathan97