jackw01/led-control #54: issu sudo command not working

PietKrobot, 2023-10-08:
hello i have no idea why my rasbpberry pi is not finding the ledcontrolmap is there annyone that can help me. And yeah sorry bit i am new to raspberry and coding and could defeneltly use som help <3
[image]

MrBurgerCZ, 2023-11-04:
Same problem here

jackw01, 2023-11-19:
I can't see the entire setup.py output so I don't know what the issue is, but try switching to the main branch (`git checkout master`) and rerunning `sudo python3 setup.py develop`

MrOwl42, 2023-11-19:
Hi Jack, 
I've had the same issue. I tried your suggestion and got an error with that command. I've attached the txt as an example of everything that comes up on my raspberry pi 3b+. 

Hope this helps! I would love to use it myself.

[led-issue1.txt](https://github.com/jackw01/led-control/files/13402895/led-issue1.txt)

MrOwl42, 2023-11-22:
Thanks for the reply! I'm not sure if I should open a new issue or not, but I upgraded to Python 3.12.0. I got a warning now with incompatible function type which ended with the same warning of needs Python 3.9 or greater.

I'm very novice at this so I tried the numpy == 1.21.0 change, and got an error with cythonize.

[led-issue2.txt](https://github.com/jackw01/led-control/files/13434386/led-issue2.txt)

Much thanks

jackw01, 2023-11-29:
Based on the output you sent, you still have an older version of python installed and `python3` is an alias for it, while `python` refers to the new python 3.12 installation. Try `sudo python setup.py develop` instead.