jackw01/led-control #55: Issue starting up

Dad-Ju, 2023-10-31:


jackw01, 2023-10-31:
Try adding `'Werkzeug==2.2.2'` to the requirements list in setup.py.

Dad-Ju, 2023-11-01:
Thanks this seamed to help :)

closed 2023-11-01 by Dad-Ju

adonaros, 2024-06-04:
one more issue, that was fixed by doing..
sudo apt-get install libopenblas-base

error was
root@raspberrypi:~/led-control# ledcontrol
Traceback (most recent call last):
  File "/root/led-control/.eggs/numpy-1.26.4-py3.9-linux-armv7l.egg/numpy/core/__init__.py", line 24, in <module>
    from . import multiarray
  File "/root/led-control/.eggs/numpy-1.26.4-py3.9-linux-armv7l.egg/numpy/core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/root/led-control/.eggs/numpy-1.26.4-py3.9-linux-armv7l.egg/numpy/core/overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/led-control/.eggs/numpy-1.26.4-py3.9-linux-armv7l.egg/numpy/__init__.py", line 130, in <module>
    from numpy.__config__ import show as show_config
  File "/root/led-control/.eggs/numpy-1.26.4-py3.9-linux-armv7l.egg/numpy/__config__.py", line 4, in <module>
    from numpy.core._multiarray_umath import (
  File "/root/led-control/.eggs/numpy-1.26.4-py3.9-linux-armv7l.egg/numpy/core/__init__.py", line 50, in <module>
    raise ImportError(msg)
ImportError:

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.9 from "/usr/bin/python3"
  * The NumPy version is: "1.26.4"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/bin/ledcontrol", line 33, in <module>
    sys.exit(load_entry_point('led-control', 'console_scripts', 'ledcontrol')())
  File "/usr/local/bin/ledcontrol", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.9/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/root/led-control/ledcontrol/__init__.py", line 5, in <module>
    from ledcontrol.app import create_app
  File "/root/led-control/ledcontrol/app.py", line 12, in <module>
    from ledcontrol.ledcontroller import LEDController
  File "/root/led-control/ledcontrol/ledcontroller.py", line 6, in <module>
    import numpy as np
  File "/root/led-control/.eggs/numpy-1.26.4-py3.9-linux-armv7l.egg/numpy/__init__.py", line 135, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
root@raspberrypi:~/led-control#