jackw01/led-control #57: No module named '_ledcontrol_rpi_ws281x_driver

devastatorx, 2023-11-23:
After installing ledcontrol and then attempting to run i get the following:

`user@abyssbar:~/led-control $ sudo ledcontrol --led_count 69
Traceback (most recent call last):
  File "/usr/local/bin/ledcontrol", line 33, in <module>
    sys.exit(load_entry_point('led-control', 'console_scripts', 'ledcontrol')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/ledcontrol", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 202, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/user/led-control/ledcontrol/__init__.py", line 5, in <module>
    from ledcontrol.app import create_app
  File "/home/user/led-control/ledcontrol/app.py", line 11, in <module>
    from ledcontrol.animationcontroller import AnimationController
  File "/home/user/led-control/ledcontrol/animationcontroller.py", line 14, in <module>
    import ledcontrol.animationfunctions as animfunctions
  File "/home/user/led-control/ledcontrol/animationfunctions.py", line 7, in <module>
    import ledcontrol.driver as driver
  File "/home/user/led-control/ledcontrol/driver/__init__.py", line 17, in <module>
    from _ledcontrol_rpi_ws281x_driver import *
ModuleNotFoundError: No module named '_ledcontrol_rpi_ws281x_driver'`

Which seems that the module is not built, I did build the module after the initial installation using `swig -python ./ledcontrol/driver/ledcontrol_rpi_ws281x_driver.i && sudo python3 setup.py develop` and it was successful.  Any ideas what might be going on here?

jayanwana, 2023-11-26:
I also have this issue on my raspberry pi 4b

jackw01, 2023-11-29:
What operating system are you running and what does `cat /sys/firmware/devicetree/base/model` return?

jayanwana, 2023-11-29:
> What operating system are you running
Raspbian GNU/Linux 11 (bullseye)
>  and what does `cat /sys/firmware/devicetree/base/model` return?
Raspberry Pi 4 Model B Rev 1.4

jayanwana, 2023-11-29:
> What operating system are you running and what does cat /sys/firmware/devicetree/base/model return?

Raspbian GNU/Linux 11 (bullseye)
Raspberry Pi 4 Model B Rev 1.4