jackw01/led-control #59: can't run setup , something to do with bjoern?

oknoote, 2023-12-28:
Sorry a bit of a noob here when it comes to python but when I run
`$ sudo python3 setup.py develop`
I get the following error followed by a stacktrace:

```
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [19 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-3.9
      copying bjoern.py -> build/lib.linux-aarch64-3.9
      running build_ext
      building '_bjoern' extension
      creating build/temp.linux-aarch64-3.9
      creating build/temp.linux-aarch64-3.9/bjoern
      creating build/temp.linux-aarch64-3.9/http-parser
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DSIGNAL_CHECK_INTERVAL=0.1 -DWANT_SIGNAL_HANDLING=yes -DWANT_SIGINT_HANDLING=yes -Ihttp-parser -Istatsd-c-client -I/usr/include/libev -I/opt/local/include -I/opt/homebrew/include -I/usr/local/include -I/usr/include/python3.9 -c bjoern/_bjoernmodule.c -o build/temp.linux-aarch64-3.9/bjoern/_bjoernmodule.o -std=c99 -fno-strict-aliasing -fcommon -fPIC -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -g
      In file included from bjoern/wsgi.h:2,
                       from bjoern/_bjoernmodule.c:3:
      bjoern/request.h:4:10: fatal error: ev.h: No such file or directory
          4 | #include <ev.h>
            |          ^~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
```

Running headless on an RPI 3B.

jackw01, 2023-12-31:
Try `sudo apt install libev-dev`

oknoote, 2024-01-09:
Hi @jackw01 ,

thanks for helping out.

I'm running into a a whole different kind of error this time unfortunately.. I upgraded to a RPI 4B and followed the installation instructions but now getting different errors running the setup.py script again. I'm running python 3.11.2.

Everything is going fine until running the setup.py script again, this time with a whole range of different errors:
```
$ sudo python setup.py develop
/usr/lib/python3/dist-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
  warnings.warn(
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [366 lines of output]
```

and then hundreds of lines of which I'll post the start here:

```
 running bdist_wheel
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -std=c++11 -mavx512f -c test-avx512.cpp -o test-avx512.o
      aarch64-linux-gnu-gcc: error: unrecognized command-line option ‘-mavx512f’
      warning: build: Compiler does not support avx512 flags: -std=c++11 -mavx512f
      
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -std=c++11 -mavx2 -c test-avx2.cpp -o test-avx2.o
      aarch64-linux-gnu-gcc: error: unrecognized command-line option ‘-mavx2’
      warning: build: Compiler does not support avx2 flags: -std=c++11 -mavx2
      
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -std=c++11 -msse4.1 -c test-sse41.cpp -o test-sse41.o
      aarch64-linux-gnu-gcc: error: unrecognized command-line option ‘-msse4.1’
      warning: build: Compiler does not support sse41 flags: -std=c++11 -msse4.1
      
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -std=c++11 -msse2 -c test-sse2.cpp -o test-sse2.o
      aarch64-linux-gnu-gcc: error: unrecognized command-line option ‘-msse2’
      warning: build: Compiler does not support sse2 flags: -std=c++11 -msse2
      
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/pyfastnoisesimd
      copying pyfastnoisesimd/__init__.py -> build/lib.linux-aarch64-cpython-311/pyfastnoisesimd
      copying pyfastnoisesimd/version.py -> build/lib.linux-aarch64-cpython-311/pyfastnoisesimd
      copying pyfastnoisesimd/test_fns.py -> build/lib.linux-aarch64-cpython-311/pyfastnoisesimd
      copying pyfastnoisesimd/helpers.py -> build/lib.linux-aarch64-cpython-311/pyfastnoisesimd
      running build_ext
      building 'pyfastnoisesimd.extension' extension
      creating build/temp.linux-aarch64-cpython-311
      creating build/temp.linux-aarch64-cpython-311/pyfastnoisesimd
      creating build/temp.linux-aarch64-cpython-311/pyfastnoisesimd/fastnoisesimd
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/lib/python3/dist-packages/numpy/core/include -Ipyfastnoisesimd -Ipyfastnoisesimd/fastnoisesimd/ -I/usr/include/python3.11 -c pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD.cpp -o build/temp.linux-aarch64-cpython-311/pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD.o -std=c++11
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/lib/python3/dist-packages/numpy/core/include -Ipyfastnoisesimd -Ipyfastnoisesimd/fastnoisesimd/ -I/usr/include/python3.11 -c pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp -o build/temp.linux-aarch64-cpython-311/pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.o -std=c++11
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/lib/python3/dist-packages/numpy/core/include -Ipyfastnoisesimd -Ipyfastnoisesimd/fastnoisesimd/ -I/usr/include/python3.11 -c pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_neon.cpp -o build/temp.linux-aarch64-cpython-311/pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_neon.o -std=c++11
      In file included from pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_neon.cpp:37:
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp: In function ‘SIMDf L5_FUNC_GradCoord(SIMDi, SIMDi, SIMDi, SIMDi, SIMDf, SIMDf, SIMDf)’:
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp:204:41: note: use ‘-flax-vector-conversions’ to permit conversions between vectors with differing element types or numbers of subparts
        204 | #define SIMDf_BLENDV(a,b,mask) vbslq_f32(mask,b,a)
            |                                ~~~~~~~~~^~~~~~~~~~
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp:781:19: note: in expansion of macro ‘SIMDf_BLENDV’
        781 |         SIMDf u = SIMDf_BLENDV(y, x, l8);
            |                   ^~~~~~~~~~~~
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp:781:38: error: cannot convert ‘MASK’ {aka ‘int32x4_t’} to ‘uint32x4_t’
[... cut ...]
        786 |         SIMDf v = SIMDf_BLENDV(SIMDf_BLENDV(z, x, h12o14), y, l4);
            |                                                   ^~~~~~
            |                                                   |
            |                                                   MASK {aka int32x4_t}

```
and so on.....

From what I can read out of this is that my gcc-compiler doesn't support mentioned flags, any other help from your end possible?

`$ aarch64-linux-gnu-gcc --version`
outputs
`aarch64-linux-gnu-gcc (Debian 12.2.0-14) 12.2.0`

Thanks...

jackw01, 2024-01-10:
Please try the latest commit in the master branch instead of the 2.1.0 tag.

oknoote, 2024-01-13:
It's working my man, thanks so much!

oknoote, 2024-01-13:
Perfect developer companion when working with rgb led strips, loving it !

closed 2024-01-27 by jackw01