jackw01/led-control #61: Error when running setup.py - deprecation warning, script not starting :(

eliphaslevi1810, 2024-05-30:
Hi everyone! Using an older Pi model with raspberry Pi OS that has been updated with Python 3.8 
I have followed the latest indicated steps for setup on a Pi, however at line #5, 

when I run "sudo python3 setup.py develop" I receive a message prompting me with the following:

/user/lib/python3/dist-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
warnings.warn(

Any idea what can be causing this? Obviously a linux noob but I am pretty sure to have correctly installed/updated the required dependencies. 

Thank you in advance for any help you can provide!

michael64w, 2024-07-11:
Had the same error using a Raspberry pi 4, I am a complete noob to, but found out its related to SIMD errors. its looking for a library called pyfastnoisesimd which the pi cant install since the compiler is based on a different chip architecture ARM if i remember. i had chat gpt rewrite the setup script. also had install multiple libraries. but then it worked. if you need it i could look for the adjusted setup script

decode40, 2024-09-03:
> Had the same error using a Raspberry pi 4, I am a complete noob to, but found out its related to SIMD errors. its looking for a library called pyfastnoisesimd which the pi cant install since the [...]

It would be amazing if you could post your script.

milhouse34521, 2024-09-26:
Same issue on a 3 B+.  Any solution? Warning about PEP, then pyfastnoisesimd installation fails (failed building wheel).

mattglg, 2024-11-10:
Looks like this has already been fixed but no version create for it - I just missed the 'git checkout tags/v2.1.1' command from the install instructions and it worked for me.

The setup.py command took ages on a PiZero but after a long time and randomly clicking enter it finished.
I made it a service using comments in this issue and it all working fine - https://github.com/jackw01/led-control/issues/3

XenonofArcticus, 2024-12-17:
I _think_ I might be having the same problem as @mattglg . I did checkout 2.1.1 per the instructions, but when I try to build on a RPi Zero W, I get errors centered around pyfastnoisesimd. I'm a developer, but this is out of my wheelhouse. Anyone have any suggestions?

@ledlights:~/led-control $ sudo python3 setup.py develop
/usr/lib/python3/dist-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
  warnings.warn(
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [998 lines of output]
      WARNING: NumPy not installed, it is required for compilation.
      running bdist_wheel
      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -std=c++11 -mavx512f -c test-avx512.cpp -o test-avx512.o
      arm-linux-gnueabihf-gcc: error: unrecognized command-line option ‘-mavx512f’
      warning: build: Compiler does not support avx512 flags: -std=c++11 -mavx512f

      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -std=c++11 -mavx2 -c test-avx2.cpp -o test-avx2.o
      arm-linux-gnueabihf-gcc: error: unrecognized command-line option ‘-mavx2’
      warning: build: Compiler does not support avx2 flags: -std=c++11 -mavx2

      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -std=c++11 -msse4.1 -c test-sse41.cpp -o test-sse41.o
      arm-linux-gnueabihf-gcc: error: unrecognized command-line option ‘-msse4.1’
      warning: build: Compiler does not support sse41 flags: -std=c++11 -msse4.1

      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -std=c++11 -msse2 -c test-sse2.cpp -o test-sse2.o
      arm-linux-gnueabihf-gcc: error: unrecognized command-line option ‘-msse2’
      warning: build: Compiler does not support sse2 flags: -std=c++11 -msse2

      running build
      running build_py
      creating build
      creating build/lib.linux-armv6l-cpython-311
      creating build/lib.linux-armv6l-cpython-311/pyfastnoisesimd
      copying pyfastnoisesimd/test_fns.py -> build/lib.linux-armv6l-cpython-311/pyfastnoisesimd
      copying pyfastnoisesimd/version.py -> build/lib.linux-armv6l-cpython-311/pyfastnoisesimd
      copying pyfastnoisesimd/helpers.py -> build/lib.linux-armv6l-cpython-311/pyfastnoisesimd
      copying pyfastnoisesimd/__init__.py -> build/lib.linux-armv6l-cpython-311/pyfastnoisesimd
      running build_ext
      building 'pyfastnoisesimd.extension' extension
      creating build/temp.linux-armv6l-cpython-311
      creating build/temp.linux-armv6l-cpython-311/pyfastnoisesimd
      creating build/temp.linux-armv6l-cpython-311/pyfastnoisesimd/fastnoisesimd
      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -Ipyfastnoisesimd -Ipyfastnoisesimd/fastnoisesimd/ -I/usr/include/python3.11 -c pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD.cpp -o build/temp.linux-armv6l-cpython-311/pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD.o -std=c++11
      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -Ipyfastnoisesimd -Ipyfastnoisesimd/fastnoisesimd/ -I/usr/include/python3.11 -c pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp -o build/temp.linux-armv6l-cpython-311/pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.o -std=c++11
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp: In function ‘int L0_FUNC_CAST_TO_INT(float)’:
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp:424:49: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
        424 | static int FUNC(CAST_TO_INT)(float f) { return *reinterpret_cast<int*>(&f); }
            |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp: In function ‘float L0_FUNC_CAST_TO_FLOAT(int)’:
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp:425:51: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
        425 | static float FUNC(CAST_TO_FLOAT)(int i) { return *reinterpret_cast<float*>(&i); }
            |                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp: In function ‘float L0_FUNC_INV_SQRT(float)’:
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp:446:18: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
        446 |         int i = *(int*)&x;
            |                  ^~~~~~~~
      pyfastnoisesimd/fastnoisesimd/FastNoiseSIMD_internal.cpp:448:14: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
        448 |         x = *(float*)&i;
            |              ^~~~~~~~~~
[... cut ...]
  File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 949, in fetch_build_egg
    return fetch_build_egg(self, req)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/setuptools/installer.py", line 84, in fetch_build_egg
    raise DistutilsError(str(e)) from e
distutils.errors.DistutilsError: Command '['/usr/bin/python3', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpm3p1zm5z', '--quiet', 'pyfastnoisesimd>=0.4.2']' returned non-zero exit status 1.
xenon@ledlights:~/led-control $